TimVosch/Aqua #17: Atmega (arduino) remote software upgrade

TimVosch, 2016-10-25:
Upgrading atmega software remotely is possible, but what happens in case of failure? And how can we recover?

SvMierlo, 2016-10-30:
https://github.com/esp8266/Arduino/issues/268
This might help your research on this matter. @TimVosch

TimVosch, 2016-10-30:
The esp8266 is a spinoff of an arduino, so the arduino could be replaced by the esp8266, however I'm not sure if we'd run in trouble with the sensors and lora equipment

TimVosch, 2016-10-30:
I was thinking about keeping the two arduino's on the board, which was shown our the first meeting. We could use one arduino to relay the LoRa module to the other, perform firmware upgrades to the other and trigger the other to power-up / down.

TimVosch, 2016-10-30:
I'm figuring out how we can use two atmega chips\* to program eachother, or atleast have a master and slave. Protocol can be found on the atmel website.

http://www.atmel.com/Images/doc0943.pdf

\* arduino is the whole board, atmega is the chip we're programming.

TimVosch, 2016-10-30:
Atmega168-20 Should be able to program itself, if it's possible for it to communicate to the LoRa module, then relay it, as well than this would be a great candidate.
It's also made for low power usage.

TimVosch, 2016-11-14:
This issue will be moved to the prototyping phase. We'll be putting several ideas to the test.

TimVosch, 2016-11-19:
Using the SPI library, that is installed by default, we should be able to do this rather easily.