TimVosch/Aqua #23: Make a working LoRaWAN network

SvMierlo, 2016-11-09:
We will need a working LoRaWAN network to see what we can do with it and how we can use it.

SvMierlo, 2016-11-09:
https://www.thethingsnetwork.org/forum/t/open-source-lorawan-network-server/1485
This guy set up his own LoRaWAN-network without interrupting the Wifi-network.

TimVosch, 2016-11-09:
Why would LoRa interrupt with the WiFi? They're operating at different frequencies

TimVosch, 2016-11-10:
Took us more than 10 hours to get an example running, but we can confirm that both the gateway as well as the module for arduino is capable of sending and receiving information. The example, named **Prototype_Lora1_Basic_Sending**, will be available in the prototypes folder.

TimVosch, 2016-11-11:
Today we've tried to send a packet from school to the gateway at my home, which is about 1 kilometer. Note that the Gateway was placed inside my room near the window 1st floor.

We could not send or receive any packet from this distance. We assume there was too much interference.

TimVosch, 2016-11-11:
I've sent a request for funds to buy different antennas as wel as some other Arduino modules, for testing purposes.

TimVosch, 2016-11-14:
Better antennas will be bought asap!